ErichStyger/mcuoneclipse #23: PE LED error on copy from KDS200 to KDS300 project

neilh10, 2015-07-07:
Image of LedRedCopied:LED from KDS200 project.
It has error on Properties.Hw Interface. Pin "Error in the inherited component setting"
On attempting to change the Pin it can't find a suitable component

[image]

If the LED is started from new it works OK
[image]

ErichStyger, 2015-07-08:
Ok, I see that problem on my end too. It seems to be related with the new Processor Expert component repositories. I have filed a bug report about this at Freescale.

ErichStyger, 2015-11-20:
I have not heard from Freescale about a resolution of this issue.

closed 2015-11-20 by ErichStyger