ErichStyger/mcuoneclipse #24: How to Your USB PE component

rongwaang, 2015-11-19:
Dear Professor Erich,

     We downloaded and added your USB PE component into PE without using RTOS.   

   When using IAR compiled the PE generated code, we got " "freeRTOS.h" not found".

     Our main project does not use any RTOS currently.

     Can the problem be fixed without using FreeRTOS? Your past note on this side seems it can. But I have no idea how.

     Thank you very much,

Rong

ErichStyger, 2015-11-20:
Hi Rong,
thanks for reporting this issue. Indeed, I should have better used the common memory utility. I have now fixed this, and it is available in this commit:
https://github.com/ErichStyger/McuOnEclipse_PEx/commit/85675b65f588f14e75809127d824b4005f3e02f2

Erich

rongwaang, 2015-11-23:
Dear Professor Erich,
    Thank you very much for your help
.
    There is another compiler error in the same file:
"RES_NOT_ENOUGH_CORE" is undefined.

     Please help,

Rong

rongwaang, 2015-11-23:
Dear Professor Erich,

rongwaang, 2015-11-23:
Dear Professor Erich,

   We do not use any RTOS. How to fix the following compiler error, which happened after comment out "#include freertos.h":

 "RES_NOT_ENOUGH_CORE" is undefined.

   Thanks,

Rong

ErichStyger, 2015-11-23:
I'm not able to reproduce that. Can you provide more details?

Thank you,
Erich

rongwaang, 2015-11-23:
Dear Professor Erich,

    Is the constant "RES_NOT_ENOUGH_CORE" defined in  freertos.h???
    The error message ""RES_NOT_ENOUGH_CORE" is undefined" shows up after deleting
   "#include freertos.h".
    The C code statement "return RES_NOT_ENOUGH_CORE" is near the place where you modified the code last time. Compiler gives error to this constant twice since it is used twice.

    We do not have RTOS and only added your "FSL_USB_STACK" and "FatfsMem_USB_MSD" components into our PE, which is part of our IAR project.

    The question is where the constant "RES_NOT_ENOUGH_CORE" is defined - Is it in  freertos.h???
    
Thanks,

Rong

ErichStyger, 2015-11-23:
RES_NOT_ENOUGH_CORE is part of diskio.h:

/* Results of Disk Functions */
typedef enum {
  RES_OK = 0,           /* 0: Successful */
  RES_ERROR,    	/* 1: R/W Error */
  RES_WRPRT,            /* 2: Write Protected */
  RES_NOTRDY,           /* 3: Not Ready */
  RES_PARERR,           /* 4: Invalid Parameter */
  RES_NOT_ENOUGH_CORE   /* 5: Not enough memory */ /* << EST added */
} DRESULT;
It is generated for me from FatFsMem_USB_MSD component.

rongwaang, 2015-11-23:
Dear Professor Erich,

    I do have diskio,h located in ....\PE\Source\, dated as April 2015 - Not generated this time. 
- "RES_NOT_ENOUGH_CORE /5" is missing in this file. There are only other 4's.
    I added the above missing line and I am able to compile the all PE files
1, I am not sure if there is any other difference between my diskio.h and your diskio.h .
2. When I tried to rebuild the while IAR project, I got:
   "Build error: Multiple tools write to the same file"
   Now I can NOT build my IAR project.
   
   I never see this error before - "Build error: Multiple tools write to the same file"
  
   Thank you very much for your help,

Rong

rongwaang, 2015-11-23:
Dear Professor Erich,

    There are more info with  "Build error: Multiple tools write to the same file"
Tool 1 - .......\PE\Generated_Code\mem_util,c
Tool 2: ........\PE\Sources\USB_Host\app_msd\mem_util.c

    Look like there is some conflict there.

    Thank you very much for your help,

Rong

rongwaang, 2015-11-23:
Dear Professor Erich,

    All files in Tool 2: ........\PE\Sources\USB_Host\app_msd\ were dated months ago.
    1. device_driver_info.c and .h
    2. mem_util.c and .h
    3. msd_commands.c and .h

    Should I delete them?

    Thanks,

Rong

ErichStyger, 2015-11-23:
Hello,
I don't know why you have these old files. So yes, best if you make a backup (just in case) and delete all the generated files, then re-generate them. It seems that the IAR tools are doing bad things with your files :-(

rongwaang, 2015-11-23:
Dear Professor Erich,

    I will try tomorrow since I will have a meeting soon.. I will let you know tomorrow what happens 
    Another person was working on this project who loaded Freescale USB component (without FAT). I suspect if those files were generated back then. I deleted that component and loaded yours. That is the history.

    Thank you so much!

Rong

rongwaang, 2015-11-24:
Dear Professor Erich,

1. I deleted all generated files. Then I rerun the PE code generation. 
2. diskio.h was not generated however (Why). I had to copy the older version of this file into the project.
3. When compiled in IAR, there is errors in khci_kinetis.c on line 306 and line 620 -
    "g_Mem" is not defined.
    I do not understand it.

   Could you help???

   Thanks,

Rong
    I do not understand

rongwaang, 2015-11-24:
Dear Professor Erich,

(Continue)    Your "FSL_USB_STACK" and "FatfsMem_USB_MSD" components are in our PE, which is part of our IAR project, Our main project does not use any RTOS.

"diskio.h" was not generated by PE
When compiled in IAR, there is errors in the generated file "khci_kinetis.c" 
""g_Mem" is not defined" on line 306 and line 620 -
.
Thanks,

Rong

ErichStyger, 2015-11-24:
Hi Rong,
I have not used IAR for a while, and not sure what is wrong. Maybe you can attach your project so I can have a look?

Erich

rongwaang, 2015-11-25:
Dear Professor Erich,

1. How to attach our project zip file and send it to you? 

2. Our PE did not generate "diskio.h" (I had to copy it from KSDK1.2.0 USB files - not 100% as same as your "diskio.h"). This problem is unrelated to IAR

3. Our PE is v10.4.2. I loaded your "FSL_USB_STACK" and "FatfsMem_USB_MSD" components without RTOS into this PE. No "diskio.h" was generated.

4."khci_kinetis.c" is PE generated file where IAR gives the compiler errors in line 306 and line 620.
 .
 Thanks,

Rong

ErichStyger, 2015-11-26:
You should be able to attach files to this ticket (at least I have an option if I post a comment).
Other than that: you can send it to my email address listed on http://mcuoneclipse.com/about/

rongwaang, 2015-11-26:
Dear Professor Erich,

    Currently our company is closed for Thanksgivings Holiday, I could not access my office computer to send you the project files. I will do next Monday (11/30) for sure..
 .
 Thanks,

Rong

rongwaang, 2015-11-30:
Dear Professor Erich,

     This place does not accept any zipped file 
("Unfortunately, we don’t support that file type.  Try again with a PNG, GIF, JPG, DOCX, PPTX, XLSX, TXT, or PDF"). 

     I could not find your e-mail address in that link. Could you send it to me directly?
    
Two issues I have now:
1. Our PE did not generate "diskio.h" (I had to copy it from KSDK1.2.0 USB files - not 100% as same as your "diskio.h"). This problem is unrelated to IAR.  Our PE is v10.4.2. 
    I loaded your  "FSL_USB_STACK" and "FatfsMem_USB_MSD" components without RTOS into this PE. But no "diskio.h" was generated.

2. "khci_kinetis.c" is PE generated file where IAR gives the compiler errors in line 306 and line 620.

   Thank you very much for your help,

Rong

rongwaang, 2015-11-30:
Dear Professor Erich,

    I found your e-mail address and sent the zipped PE project file to your e-mail address. Our PE is v10.4.2

   Our entire IAR project file is too big to send. I just sent you our zipped PE files first.

   I will try to send the entire zipped project file later if PE files are not enough for you to fix the problem.

   Please let me know if you receive and what else you need.
      
Thank you very much for your help,

Rong

ErichStyger, 2015-12-02:
I see now the problem.
The wrong name of the buffer is used by the FSL stack for IAR.
I have made a fix here:
https://github.com/ErichStyger/McuOnEclipse_PEx/commit/0c206f47338033d59ad3e72373d8d69141820652
You can apply this fix to your own version of the component. The file affected should be here on your machine:
C:\ProgramData\Processor Expert\PEXDRV_PE5_3\Drivers\FSL_USB_Stack\Host\source\driver\kinetis\khci_kinetis.c
So you can do the patch manually.
Then I was able to compile your project, but not able to link it because not all the needed sources are present in the zip file.

I hope this helps,
Erich

rongwaang, 2015-12-02:
Dear Professor Erich,

   Thank you so much for your timely help. 

   Now I am able to compile the PE and the whole project. 

    Our PE did not generate "diskio.h",  I copied it from KSDK1.2.0 USB files and added one line code. Hope you can verify its correctness - it is included in the files I sent.

   Our USB application is rather simple - The USB device just occasionally reads one, only one, fixed file written by our firmware. Do you think it is going to work without RTOS? Our project does not use RTOS at all.

   Thanks again, I will test it out after write the interface.

Rong

rongwaang, 2015-12-02:
Dear Professor Erich,

1. Is there any document specifying the interface between your USB code and our application?
2. Which USB initialization functions we must call when power up from our application? 
3. Should our application main loop call "FsMSD1_AppTask()"?

   Thank you very much,

Rong

ErichStyger, 2015-12-02:
Please have a look at existing FatFS USB MSD Host examples, e.g. https://github.com/ErichStyger/mcuoneclipse/tree/master/Examples/KDS/FRDM-K22F/FRDM-K22F_USB_MSD_Host
(that one is using FreeRTOS, but does not matter: you can do the same bare metal).
As documentation, see the Freescale USB stack v4.1.1 documentation (http://cache.freescale.com/files/microcontrollers/doc/support_info/USB_STACK_RELEASE_NOTES_V4.1.1.pdf and http://www.freescale.com/webapp/sps/download/license.jsp?colCode=USB_STACK_V4.1.1&location=null).

Diskio.h is generated by Fat_FileSystem component, you need to add this one so you have a file system.

rongwaang, 2015-12-02:
Dear Professor Erich,

"Diskio.h is generated by Fat_FileSystem component, you need to add this one so you have a file system." - 
    You can see we have "FatFsMem_USB_MSD" component in our PE. 
    Should we add "Fat_FileSystem component" as well, or replace  "FatFsMem_USB_MSD" with "Fat_FileSystem component"? What else we need to add? This is very critical.

   Thank you very much,

Rong

rongwaang, 2015-12-02:
Dear Professor Erich,

   Another question: To do what we need (R/W one fixed file), how many USB components (Minimum)in your LIB we need to add in our project???
   Currently we have "FSL_USB_STACK" and "FatfsMem_USB_MSD" components. When I tried to add "Fat_FileSystem component", I got a lot of syntax error.
   
Thank you very much,

Rong

rongwaang, 2015-12-02:
Dear Professor Erich,

   (Continue) When I said "I got a lot of syntax error When I tried to add "Fat_FileSystem component"", I meant the errors are generated by PE, not IAR, 
    You have our PE project files. You may see them using PE v10.4.2.

    Thank you very much for your help,

Rong

ErichStyger, 2015-12-03:
Hi Rong,
Could you use the Kinetis Design Studio project () as a reference, e.g. this project here:

I ask because you need to carefully configure the USB clock (which was not the case in the project you had sent to me).
You need:
- FSL_USB_Stack with FSL_USB_MSD_Host as sub component
- FatFsMem_USB_MSD to handle the USB MSD interaction with FatFS
- FAT_FileSystem as imlementation of the File System with FatFsMem_USB_MSD as Drive0

I hope this helps,
Erich

rongwaang, 2015-12-03:
Dear Professor Erich,

  Our CPU PE configuration already set USB clock as 48MHz in our application project, which you have.

  "Could you use the Kinetis Design Studio project () as a reference, e.g. this project here:"
 
  Which one? Please advise.
   
  When load the third component "•FAT_FileSystem " into our PE, we got a lot of syntax errors in PE. You have our PE project and you will get the same syntax errors when load the  "•FAT_FileSystem ". Maybe this component also try to initialize the clock, which has be done by our CPU PE component?

Thank you very much for your help,

Rong

ErichStyger, 2015-12-04:
Here are projects:
https://github.com/ErichStyger/mcuoneclipse/tree/master/Examples/KDS/FRDM-K20D50M/FRDM-K20_MSD_Host
https://github.com/ErichStyger/mcuoneclipse/tree/master/Examples/KDS/FRDM-K64F120M/FRDM-K64F_USB_MSD_Host
https://github.com/ErichStyger/mcuoneclipse/tree/master/Examples/KDS/FRDM-K22F/FRDM-K22F_USB_MSD_Host

And only the USB Init will configure the USB clock, but you need to have as well the clock configured in the CPU component. I suggest that you have a look at the above projects.

I hope this helps,
Erich

rongwaang, 2015-12-04:
Dear Professor Erich,

   I will do my homework by studying the projects.

Thank you very much for your help,

Rong

closed 2015-12-14 by ErichStyger

rongwaang, 2015-12-14:
Dear Professor Erich,

    I have been busy on other work. Now I am back.
    1. Are the links below are PE based USB application projects using your components?
    2. How to download the zipped projects?
    Thanks,

Rong

Here are projects:
https://github.com/ErichStyger/mcuoneclipse/tree/master/Examples/KDS/FRDM-K20D50M/FRDM-K20_MSD_Host
https://github.com/ErichStyger/mcuoneclipse/tree/master/Examples/KDS/FRDM-K64F120M/FRDM-K64F_USB_MSD_Host
https://github.com/ErichStyger/mcuoneclipse/tree/master/Examples/KDS/FRDM-K22F/FRDM-K22F_USB_MSD_Host

ErichStyger, 2015-12-15:
1. Yes, they are using McuOnEclipse projects
2. There is a 'download zip' button on the root of every repository page: go to https://github.com/ErichStyger/mcuoneclipse and there is https://github.com/ErichStyger/mcuoneclipse/archive/master.zip

rongwaang, 2015-12-16:
Dear Professor Erich,

    I did download the zip file. There are two projects: "FRDM-K20D50M" and "FRDM-KL25Z" under IAR. I do not see "FRDM-K64F_USB_MSD_Host".

    Is this zip file the right and only file I need under IAR?
   
Thanks,

Rong

rongwaang, 2015-12-16:
Dear Professor Erich,

    I used the link in your mail to down load the zipped file.
    But when I viewed the project's PE project (Under Example Project IAR directory), I do not see any USB component there (I expect three USB components).

    May be I did not get the right zip file?

    Thank you very much for your help,

Rong

rongwaang, 2015-12-16:
Dear Professor Erich,

    I was following your KDS example PE project to import and configure your 3 USB components in my PE project. So far I am able to compile the project under PE successfully. I will try to compile it under IAR later.

    Thanks,

Rong

rongwaang, 2015-12-23:
Dear Professor Erich,

   I am able to configure and compiler the project now.

   I added   "i = FAT1_Init();" and "i = FsMSD1_HostInit();" into my initial routine in my main project.
   Also I added FsMSD1_AppTask(); into my main loop in my main project (main.c)

   However when I plug USB into my prototype board, I expect the program breaks out in the break point set in (FsMSD1_AppTask(); USB_DEVICE_ATTACHED). But it breaks out in below (cpu.c) without any break point set.

PE_ISR(Cpu_Interrupt)
{
  /* This code can be changed using the CPU bean property "Build Options / Unhandled int code" */
  /* asm (HALT) */
}

    Could you advise me how to fix it? What I missed here?

    Thank you very much and wish you Merry Christmas and Happy Holidays,

Rong

ErichStyger, 2015-12-23:
Hello Rong,
are you still using IAR for this? I suggest that you start with Kinetis Design Studio and the project as on GitHub so you have a working environment. Once you have that working, you can consider doing this with IAR. Because otherwise there are far too many variables and things which can go wrong with the IAR environment.

As for the unknown interrupt: See the following post and how to enable interrupt handlers so you know which interrupt is causing the problem for you:
http://mcuoneclipse.com/2012/02/20/oh-my-an-interrupt/

Merry Christmas and Happy Holidays,
Erich

rongwaang, 2015-12-23:
Dear Professor Erich,

   After I changed to "Own handler for every" in CPU component configuration in PE, the fault is caused by "Cpu_ivINT_Hard_Fault"

   Could it be caused by some task stack size defined too low and it needs to be increased? How can I trace it from here for "Cpu_ivINT_Hard_Fault"?

   I will try to use KDS.

   Thanks,

Rong

ErichStyger, 2015-12-23:
Hi Rong,
I have a few helpers and articles around hard faults here:
http://mcuoneclipse.com/2012/11/24/debugging-hard-faults-on-arm-cortex-m/
http://mcuoneclipse.com/2015/07/05/debugging-arm-cortex-m-hard-faults-with-gdb-custom-command/
http://mcuoneclipse.com/2012/12/28/a-processor-expert-component-to-help-with-hard-faults/

I hope this helps,
Erich

rongwaang, 2015-12-29:
Dear Professor Erich,

   I found "OTG_BUILD" is not defined in the generated code, and there is no code inside 
USB_ISR(void). That may be why CPU Interrupt Error occurs? Should "OTG_BUILD" be defined?
   Otherwise there should be the code inside USB_ISR(void). Right?   Thanks Rong

Refer to:
#ifdef OTG_BUILD
void  USB_ISR_HOST(void)
#else
	void  USB_ISR(void)
#endif

ErichStyger, 2015-12-29:
Yes, that's correct: OTG_BUILD does not have to be defined.
The USB interrupt function is defined with
	void  USB_ISR(void)
in khci_kinetis.c, line 1093.

I hope this helps,
Erich

rongwaang, 2015-12-29:
Dear Professor Erich,

    Thanks,.
    I believer before the CPU Interrupt Error occurs (Whenever USB sticker
is plugged in), the program was running in USB_ISR(void) first. I have to
set the break point there to trice where the exception occurs and why.
Right?

    Since the office is closed, I have to wait until 2016.

   Thank you for your help,
   Happy New Year!

Rong

rongwaang, 2016-01-13:
Dear Professor Erich,

   I just came back to this task today.
   Once the USB attachment is detected, the program does break in "USB_ISR".
   However the program got into "WAIT1_Waitms(delay);" listed at the end of this comment. Why???
   "void time_delay(uint_32 delay)" may cause the CPU interrupt error. 
    Should "void time_delay(uint_32 delay)" be used?
Thank you very much for your help,

Rong
   

void time_delay(uint_32 delay) {
#if 1 /* too inaccurate? */
  WAIT1_Waitms(delay);
#else
  extern volatile long delay_count; /* needs to be incremented in Events.c by a 1 ms timer */

  delay_count = 0;
  while(delay_count<delay) {
    /* wait */
  }
#endif
}
   

   Thank you for your help,

Rong

ErichStyger, 2016-01-15:
Yes, that version works for me. But you are free to use any other method for the delay if you want.

Best regards,
Erich

erdemkose78, 2016-01-15:
eric can you delete me from the mailing list

*Electronics Eng.*
*Erdem KÖSE*

ErichStyger, 2016-01-15:
I cannot do that. You have unsubscribe yourself I think.

rongwaang, 2016-01-15:
It was from "erdemkose78"

rongwaang, 2016-01-15:
Dear Professor Erich,

I am trying to trace where the CPU interrupt fault ("PE_ISR(Cpu_ivINT_Hard_Fault)") came from by following the link 
http://mcuoneclipse.com/2012/11/24/debugging-hard-faults-on-arm-cortex-m/

I insert the code below as instructed by the above link:

_attribute__((naked))
PE_ISR(Cpu_ivINT_Hard_Fault)
{
  __asm volatile (
    " movs r0,#4       \n"
    " movs r1, lr      \n"
    " tst r0, r1       \n"
    " beq _MSP         \n"
    " mrs r0, psp      \n"
    " b _HALT          \n"
  "_MSP:               \n"
    " mrs r0, msp      \n"
  "_HALT:              \n"
    " ldr r1,[r0,#20]  \n"
    " bkpt #0          \n"
  );
}

However IAR gives the compiler errors on "__attribute__((naked));"

Error[Pe079]: expected a type specifier C:\Doc\USB\USB-PC_Demo\PC_GUI_DEMO\Source\InputOutputPCB\Application\PE\Generated_Code\Cpu.c 153 
Error[Pe260]: explicit type is missing ("int" assumed) C:\Doc\USB\USB-PC_Demo\PC_GUI_DEMO\Source\InputOutputPCB\Application\PE\Generated_Code\Cpu.c 153 
Error while running C/C++ Compiler 

Do you know how why? 

Thanks,

Rong

rongwaang, 2016-01-15:
Dear Professor,

   I use "__irq __arm PE_ISR(Cpu_ivINT_Hard_Fault);" to replace __attribute__((naked)) under IAR.
   But I found the asm code syntax may be different also under IAR.

   Thanks,

Rong

ErichStyger, 2016-01-15:
The Processor Expert component should work for IAR too. Have you tried that?

rongwaang, 2016-01-15:
Dear Professor,

   I am using PE under IAR. When USB is inserted, It breaks in USB_ISR(void). However it went to 
PE_ISR(Cpu_ivINT_Hard_Fault) shortly,

   I try to add the assembly code in PE_ISR Hard_Fault to trace where it came from before getting into the Hard_Fault. I got the compiler errors on the assembly code copied from the link as below:

Error[Og010]: Inline assembler instruction has more than one size: "        beq ??_MSP         ". Use size specific syntax 
Error[Og010]: Inline assembler instruction has more than one size: "        b ??_HALT          ". Use size specific syntax 

     Thanks,
Rong

Thanks,

Rong

ErichStyger, 2016-01-15:
It does work with the Kinetis Design Studio IDE and compiler, correct? It might be a problem with the IAR compiler. I'm affraid that I won't be able to help you much (if any) for IAR because this is not what I can use :-(

rongwaang, 2016-01-15:
Dear Professor Erich,

    We do not use KDS. The code was copied from the links you gave.
    The only problem is in the branch statement. IAR does not like the two branches. Other syntax seems OK if I get rid of the branches. However I need to find out how to write the branch under IAR.

    I will ask IAR for help.

   Thanks,

Rong

rongwaang, 2016-01-15:
I use "__irq __arm PE_ISR(Cpu_ivINT_Hard_Fault);" to replace  "_attribute__((naked))"
    It seems OK now.

    However IAR compiler does not like the branch statement below

  __asm volatile (
    " movs r0,#4       \n"
    " movs r1, lr      \n"
    " tst r0, r1       \n"
    " beq _MSP         \n"
    " mrs r0, psp      \n"
    " b _HALT          \n"
  "_MSP:             \n"
    " mrs r0, msp      \n"
  "_HALT:            \n"
    " ldr r1,[r0,#20]  \n"
    " bkpt #0          \n"  
  ); 

Error[Og010]: Inline assembler instruction has more than one size: "        beq ??_MSP         ". Use size specific syntax 
Error[Og010]: Inline assembler instruction has more than one size: "        b ??_HALT          ". Use size specific syntax