ErichStyger/mcuoneclipse #26: FreeRTOS: xTaskCreate() doesn't return Task Handle

r-ben, 2016-01-07:
After updating to recent PEComponents from 2015-11-22, xTaskCreate() doesn't return the task handle any more. Maybe it's a bug in FreeRTOS 8.2.3.

Reading the task handle with xTaskGetCurrentTaskHandle() from inside the task is working.

ErichStyger, 2016-01-07:
Hello,
it works as expected for me and returns the task handle with the last parameter (address of task handle passed). Maybe you did something wrong? Below is an example code I use:

static TaskHandle_t handle;
  if (FRTOS1_xTaskCreate(BleUartTask, "BleUart", configMINIMAL_STACK_SIZE+100, NULL, tskIDLE_PRIORITY+1, &handle) != pdPASS) {
    for(;;){} /* error */
  }

I hope this helps,
Erich

r-ben, 2016-01-08:
Hello Erich,
Yes, indeed! However, the '&' for the handle disappeared in my code.

Many thanks for your fast response,
Rolf

closed 2016-01-08 by r-ben