ErichStyger/mcuoneclipse #27: FreeRTOS component fails to reload

nlem, 2016-02-18:
Hi Erich,
I have the latest set of pex components from sourceforge (5th feb). I am developing on kinetis K10 using KDS.
It all seems to work great until I shut down KDS.
When I restart KDS it is unable to load the list of components.

I am using the FreeRTOS component which also imports the Utility Component.
The only way I can get processor expert components back is to manually delete the Freertos bean from the pe file in the project.

There is an error generated when the components fail to load.
It look like this:
java.lang.NoSuchMethodError: com.processorexpert.core.repositories.PartialComponentReference.matches(Lcom/processorexpert/core/repositories/IComponentReference;Z)Z

Can you help me?

Thanks,
Nick

ErichStyger, 2016-02-18:
Hi Nick,
that's strange, never seen this. Do you have Kinetis SDK components enabled/present? Can you check (maybe disabel) component repositories? Go to Help > Preferences > Processor Expert > 
Repositories and Paths and disable the ones you don't need?
Let me know if this helps.
If not, can you provide me your .pe file so I can try it on my side?
Erich

ErichStyger, 2016-02-18:
The other idea:
shut down KDS, delete C:\ProgramData\Processor Expert\PECache and see if this helps?

nlem, 2016-02-18:
Components look like this:
[image]
The KSDK components are installed but I don't think I'm using any of them?
The KSDK part of Freertos is disabled.
I think it breaks whether it's enabled or not.
Tried deleting the PECache and disabling repositories and it didn't seem to fix it.

Should also have mentioned that I updated KDS a few days ago to the latest version.

PE file is linked below:
[ProcessorExpert.txt](https://github.com/ErichStyger/mcuoneclipse/files/136389/ProcessorExpert.txt)

ErichStyger, 2016-02-18:
Your screenshot is too small to read it :-(.
The Kinetis SDK components/repositories do come with a FreeRTOS component with the same name. I was thinking that there could be a naming conflict, that's why I think disabling the Kinetis SDK component repositories could make a difference.
Updating KDS should not be the problem (I have an updated version too).

nlem, 2016-02-18:
Tried with the SDK repo's removed.
All the SDK components went away, but doesn't seem to fix it.

If it's any help, when it's loading the components it initially shows the correct list (as above) and then the list goes away and it prints the error I mentioned.

ErichStyger, 2016-02-18:
If you move the mouse over the FreeRTOS component, it shows you the repository. Is it the same as for the others (like the utility)?
[image]

nlem, 2016-02-18:
Looks just like yous except the repository is "My Components".
Also tried renaming the Freertos component in the SDK repositories just in case and that didn't help.

ErichStyger, 2016-02-18:
Can you disable all your other repositories? Like this:
[image]
Even if you don't use SDK repositories, components from it might still be reloaded (I think)

nlem, 2016-02-18:
Already tried it.
Whole bunch of components went away but still doesn't load.
Does that PE file look right to you?

I think my next step might be to start a new project from scratch with just FreeRTOS in it.
See if I can track down where it goes wrong.

[image]

ErichStyger, 2016-02-18:
FYI, your .pe file works for me. So I still think there must be something with your other repositories?

nlem, 2016-02-18:
OK.
if it works for you it must be something in the setup here.
I'll muck around with it some more.

Thanks for you help.

nlem, 2016-02-26:
Not sure it's any help but I get the same error from the SD card component when reloading the project.

nlem, 2016-02-26:
Ended up re installing KDS and PEX.
Problem went away.
Thanks for your help anyway.

ErichStyger, 2016-02-26:
ok, good to hear that the problem is solved.

closed 2016-02-26 by ErichStyger