ErichStyger/mcuoneclipse #29: CodeWarrior 10.6 for MC9S08 C compiler don't have stdbool.h and stdint.h

phantomgz, 2016-04-20:
Hi Erich,
As mention in title, My project was very old, target CPU is MC9S08xx, running with FreeRTOS component, after install "Components 2016-04-03.zip", new version ask a new componet "KinetisSDK",
after some fixed, PE running well, but when build my project, compiler complaint "stdbool.h" "stdint.h" not found in "KSDK1.h" , and same problem also appen in "port.c"

Could you help to fixed this problem.

Thanks,
Cai

ErichStyger, 2016-04-20:
Hi Cai,
Thanks for reporting. I admit that I have not used the S08 for a while. I'll have a look and report back.
thanks,
Erich

ErichStyger, 2016-04-22:
Hi Cai,
because the S08 (and S12) compiler do not support C99 (they only support C89), including stdint.h and stdbool.h does not work :-(. I have (hopefully) fixed this in the following commit:
https://github.com/ErichStyger/McuOnEclipse_PEx/commit/1ed50d15ea3c7a34c4de5c141dc484df40bc7daf

The fix is basically guarding those includes with checking if __HIWARE__ is defined (it is defined for the S08 and S12 compiler).
If you cannot wait for the next release of the components (no date yet), you could simply add dummy/empty stdbool.h and stdint.h as a workaround.

Thanks again for reporting,
Erich

closed 2016-04-22 by ErichStyger