ErichStyger/mcuoneclipse #30: InitUSB error with MKL26 with KSDK1.3

paloky, 2016-04-28:
The Component "FSL_USB_Stack", have an error with inherit component InitUSB  with KSDS1.3.

ErichStyger, 2016-04-28:
Do you mean the Kinetis SDK v1.3? If so, yes, these components are not compatible with the Kinetis SDK. You need to create a project *without* the Kinetis SDK if you want to use the FSL_USB_Stack component.

ErichStyger, 2016-04-28:
Accidentially closed the issue, sorry. I'll keep it open for now. But to me this is a problem of the Kinetis SDK, not of the component :-(

paloky, 2016-04-28:
Hi Erich.
Thanks for your fast response.
Can you tell me, (If it's possible) how can change the project propierties to don't use the SDK ?
Thanks a lot.

ErichStyger, 2016-04-28:
You cannot change an existing project not to use the Kinetis SDK as far as I know.
You need to create a new project. There is a dialog while creating the project about which SDK to use. Select 'none' as SDK.
I hope this helps?

paloky, 2016-04-28:
Thanks a lot.
I solved with other "non ortodox form".  I desinstall Eclipse SDK1.3 plugin, next I add FSL_USB_Stack component, and next I reinstall the SDK1.3.
It seem work OK now.
Thanks again.

ErichStyger, 2016-04-28:
ok, great!

closed 2016-04-28 by ErichStyger