ErichStyger/mcuoneclipse #31: CriticalSection bug

fbstj, 2016-05-06:
I've found a bug that I've [reported on the forum](https://mcuoneclipse.com/2016/04/03/mcuoneclipse-components-3-apr-2016-release/#comment-82602) I thought I'd report it here too, since it'll look nicer 

Updating CriticalSection (CS1) to 1.008 seems to cause the SDHC sub-component of my FAT_FileSystem/FatFsMemSDHC break. it somehow decides that sdhc1_SetWriteProtecton is ill-defined??

```
‘Building file: ../Generated_Code/SDHC1.c’
‘Invoking: Cross ARM C Compiler’
arm-none-eabi-gcc -mcpu=cortex-m4 -mthumb -O0 -fmessage-length=0 -fsigned-char -ffunction-sections -fdata-sections -g3 -I”C:/Code/kinetis/SF4V/Static_Code/PDD” -I”C:/Code/kinetis/SF4V/Static_Code/IO_Map” -I”C:/Code/kinetis/SF4V/Sources” -I”C:/Code/kinetis/SF4V/Generated_Code” -std=c99 -MMD -MP -MF”Generated_Code/SDHC1.d” -MT”Generated_Code/SDHC1.o” -c -o “Generated_Code/SDHC1.o” “../Generated_Code/SDHC1.c”
../Generated_Code/SDHC1.c:1474:12: error: conflicting types for ‘SDHC1_SetWriteProtection’
LDD_TError SDHC1_SetWriteProtection(LDD_TDeviceData *DeviceDataPtr, LDD_SDHC_TWriteProtectType Type, uint32_t Address, bool Protected)
^
In file included from ../Generated_Code/FATM1.h:64:0,
from ../Generated_Code/SDHC1.c:129:
../Generated_Code/SDHC1.h:597:12: note: previous declaration of ‘SDHC1_SetWriteProtection’ was here
LDD_TError SDHC1_SetWriteProtection(LDD_TDeviceData *DeviceDataPtr, LDD_SDHC_TWriteProtectType Type, uint32_t Address, bool Protected);
^
make: *** [Generated_Code/SDHC1.o] Error 1
```

the problem is here:

```
-/* Include shared modules, which are used for whole project */
-#include “PE_Types.h”
-#include “PE_Error.h”
-#include “PE_Const.h”
-#include “IO_Map.h”
/* Include inherited beans */
+#include “KSDK1.h”

-#include “Cpu.h”
–
+#if KSDK1_SDK_VERSION_USED == KSDK1_KSDK1_SDK_VERSION_NONE
+/* Include shared modules, which are used for whole project */
+ #include “PE_Types.h”
+ #include “PE_Error.h”
+ #include “PE_Const.h”
+ #include “IO_Map.h”
+ #include “Cpu.h”
+#endif
```

`KSDK1_KSDK1_SDK_VERSION_NONE` has an extra `KSDK1_`

ErichStyger, 2016-05-06:
Hello,
thanks for reporting. Indeed, that was a copy-paste error :-(.
I have found this one too (the same copy-paste mistake is as well in other components :-(). I have it already fixed on GitHub (https://github.com/ErichStyger/McuOnEclipse_PEx/blob/master/Drivers/sw/CriticalSection.drv), but I had no time yet to release the new components. If time permits, I'll release them this weekend.
Erich

ErichStyger, 2016-05-08:
I have released a new version of the components which has this fixed.
Thanks again for reporting it!
Erich

fbstj, 2016-05-09:
Thank you for the quick fix! this set of components is so useful you're doing such a great service to us users of these chips!

fbstj, 2016-05-09:
OK after upgrading all the components to the latest releases (CriticalSection still reports as 1.008 even tho it's changed since the last but *shrugs*) but I'm still getting the error on SDHC1

fbstj, 2016-05-09:
Reverting various changes until it works, I find that updating the Timeout component from 1.030 to 1.035 also breaks things

ErichStyger, 2016-05-09:
I have not seen so far issues with that new Timeout component :-(
Any details you could provide?

fbstj, 2016-05-09:
[image]

is there any other numbers that would be useful? any hints as to where I should look for more useful info than version numbers? I'm using an MK60DN512VMD10 here's the components view 

[image]

they're all updated to the 2016-05-08 release, except the timeout and cs1 which are at 2015-10-17 (using your multi-repository method) if I try to change either to 2016-05-08 (or change the CS to 2016-04-03) it fails to compile with the above error around SDHC1_SetWriteProtection (changing the timeout to 2016-04-03 doesn't cause any problems.

the only differences I can see between the generated code are related to those `+#if KSDK1_SDK_VERSION_USED == KSDK1_SDK_VERSION_NONE` blocks around the header includes, the CS1's were added and then fixed, the utils ones were fixed for the latest release.

ErichStyger, 2016-05-09:
Ok, I'm able to reproduce the problem :-(. The issue is because with the SDK changes different declaration of type 'bool' could happen :-(.
I have fixed it that way:
https://github.com/ErichStyger/McuOnEclipse_PEx/commit/5ddff476d34b6058febbeaab6e434891b461a29d
I think you should be able to make that change in the file you have installed on your end? Can you try if this solves your problem too?

Thanks,
Erich

fbstj, 2016-05-10:
That does seem to have fixed error. thanks for all your help !! now all my components are on the latest releases.

ErichStyger, 2016-05-10:
Thanks for the confirmation!

closed 2016-05-10 by ErichStyger