ErichStyger/mcuoneclipse #32: FreeRTOS compile issue

Erdyke, 2016-06-27:
Running FreeRTOS bean on Codewarrior (MCU) for DEMO09S08DZ60 board.  Bean settings are default.

1) Looks like function "UBaseType_t FRTOS_AppGetRuntimeCounterValueFromISR(void)" is being re-declared because of a different return value type.

2) Also enabling Custom portBASE_TYPE seems to totally mess up the compiler in that I get error "C1067 missing decl specifiers" and error "C2450 Expected ;" at Line 181: "BaseType_t xPortStartScheduler( void ) PRIVILEGED_FUNCTION;" in "portable.h"

ErichStyger, 2016-06-27:
I have not used it for S08 for a while. So it could be that something is broken. I'll have a look and report back.

ErichStyger, 2016-06-27:
Hi Erdyke,
Indeed, things are broken for S08 if using a dedicated timer :-(.
I have commited a fix here: https://github.com/ErichStyger/McuOnEclipse_PEx/commit/b55ffa579e596494b1dd2396cb4e9f532e6618b1
Can you check if this solves the problem on your side as well?

I appologize for the issue,
Erich

Erdyke, 2016-06-27:
Will do,

Regards,
Eric Dyke

ErichStyger, 2016-06-27:
Hi Eric,
I have fixed as well the secone issue with custom base types:
https://github.com/ErichStyger/McuOnEclipse_PEx/commit/7362338383bf2912081df7fdfa026b12c48b8a7c

Sorry as well for this issue, I have never used custom base types :-(.

Let me know if this is fixed on your side too so I can close this issue.

Thanks,
Erich

Erdyke, 2016-06-27:
Thanks. I will need to get to it tonight.

Erdyke, 2016-06-28:
The first fix allows things to compile fine.  I have not tested the other fix.
I'm pretty new to FreeRTOS so I will need to play around with it on my demo board to see how it runs with the default settings.  Thanks for all the work you do!

ErichStyger, 2016-06-29:
Thanks, and you are welcome. I'm closing the issue now. Be free to report if there are any problems or if there is anything to reopen the issue.

closed 2016-06-29 by ErichStyger

Erdyke, 2016-06-30:
Perhaps I'm missing something, but I cannot get a regular LED demo to run using default settings with a task delay of 500 ms.  Do I need to do anything with the events created from the bean?

ErichStyger, 2016-06-30:
No, nothing special to set up, and no special event handling needed. Below some code:

static void AppTask(void* param) {
  for(;;) {
      LED1_Neg();
     vTaskDelay(pdMS_TO_TICKS(500));
  }
}

  if (xTaskCreate(AppTask, "App1", configMINIMAL_STACK_SIZE, NULL, tskIDLE_PRIORITY, NULL) != pdPASS) {
    for(;;){} /* error case only, stay here! */
  }
  vTaskStartScheduler(); /* start the RTOS, create the IDLE task and run my tasks (if any) */

Erdyke, 2016-06-30:
As before, it looks like the tasks are getting created okay and are executed.  I can set a breakpoint at the LED1_NegVal(); and see we get there each time.  But there doesn't seem to be any real delay.  I'm instantly returned to the BP after continuing.

Erdyke, 2016-06-30:
In case it's useful, I uploaded my project: https://github.com/Erdyke/FreeRTOS-DEMO

ErichStyger, 2016-06-30:
Yes, that's useful. I don't have that S08 device, but I think I have found your issue: you have the Software Interrupt not properly configured. It is an older article, but the settings for FreeRTOS still apply, have a look at https://mcuoneclipse.com/2012/06/28/tutorial-freertos-on-demojm/. The swi needs to be configured properly (https://mcuoneclipse.files.wordpress.com/2012/06/assigning-software-interrupt-vector.png?w=584).
The other thing is: your heap is set to 2KByte. That's fine for a few tasks. You might increase that one later.
I hope this helps,
Erich

Erdyke, 2016-06-30:
That's the fix. Thanks for the help. Much Appreciated!