ErichStyger/mcuoneclipse #33: nRF24 project with KDS

AdyiPool, 2016-09-11:
Hello Eirch,

First of all, thank you so much for the amazing blog posts and this GitHub repo. I've been referring to you blog a lot to get up and running with my K64F board.

I want to try your nRF24 project with KDS 3.2.0, can you please give me some hint as to how to import the project into KDS? I don't see a .wsd file in the example folder, nor does KDS accent the .pe files. I'm very new to KDS in general so forgive me if I'm overlooking something obvious.

Thanks,
Ady

AdyiPool, 2016-09-11:
If I import an existing project, I can actually build and run the NeoPixel project, but something is wrong with the nRF14 project. Here's the error I get

```
11:03:15 **** Incremental Build of configuration Debug for project FRDM-K64F_nRF24 ****
make -j3 all 
Static_Code/System/subdir.mk:23: *** target pattern contains no `%'.  Stop.

11:03:15 Build Finished (took 69ms)

```

and here's line 23 in `subdir.mk`

```
# Each subdirectory must supply rules for building sources it contributes
Static_Code/System/CPU_Init.o: /Applications/KDS_v3.app/Contents/eclipse/C:/nxp/KDS_3.2.0/eclipse/ProcessorExpert/Repositories/Kinetis_Repository/lib/Kinetis/pdd2/MK64FN1M0LL12/system/CPU_Init.c
	@echo 'Building file: $<'
	@echo 'Invoking: Cross ARM C Compiler'
	arm-none-eabi-gcc -mcpu=cortex-m4 -mthumb -mfloat-abi=hard -mfpu=fpv4-sp-d16 -O0 -fmessage-length=0 -fsigned-char -ffunction-sections -fdata-sections  -g3 -I"/Applications/KDS_v3.app/Contents/eclipse/ProcessorExpert/lib/Kinetis/pdd/inc" -I"/Applications/KDS_v3.app/Contents/eclipse/ProcessorExpert/lib/Kinetis/iofiles" -I"/Users/adityatannu/workspace3.kds/FRDM-K64F_nRF24/Sources" -I"/Users/adityatannu/workspace3.kds/FRDM-K64F_nRF24/Generated_Code" -I"/Applications/KDS_v3.app/Contents/eclipse/ProcessorExpert/Repositories/Kinetis_Repository/lib/Kinetis/pdd2/MK64FN1M0LL12/system" -std=c99 -MMD -MP -MF"$(@:%.o=%.d)" -MT"$@" -c -o "$@" "$<"
	@echo 'Finished building: $<'
	@echo ' '
```

ErichStyger, 2016-09-12:
Hi Ady,

Have you installed already the McuOnEclipse Processor Expert components from SourceForge (https://sourceforge.net/projects/mcuoneclipse/files/PEx%20Components/)?

And do not forget to generate code with Processor Expert (context menu on the .pe file in Eclipse).

Let me know if you need any further assistance.

 

Erich

ErichStyger, 2016-09-12:
Hi Ady,

There are no .wsd files, as they are non-standard and require a special plugin.

You can import the projects with the normal Eclipse method (File > import > existing projects into workspace.

 

Erich

closed 2016-09-26 by ErichStyger