ErichStyger/mcuoneclipse #34: Bad cast in gprof for arm

ptrebuc, 2017-01-06:
I try to reuse in a somewhat different context your great work on gprof, and I found a bug in the git hub implementation.

gmon.c:267 frompcindex should not be used but a fresh _uint16t *_ should

p->froms is a u_short *

best regards

ErichStyger, 2017-01-09:
I'm not really sure what you mean? Are you referring to:
https://github.com/ErichStyger/mcuoneclipse/blob/59e5aeffe5d42874f33a9ea5a86669ac7e0958ce/Examples/KDS/FRDM-K64F120M/FRDM-K64F_Profiling/Profiling/gmon.c
?

ptrebuc, 2017-01-09:
> 59e5aeffe5d42874f33a9ea5a86669ac7e0958ce/Examples/KDS/FRDM-
> K64F120M/FRDM-K64F_Profiling/Profiling/gmon.c

yes
line 269
                  *frompcindex = toindex;
at this moment frompcindex contains the address of an element of p->froms.

p->froms is a ushort *
this means that if two function calls are too near the data are corrupted

Am I wrong ?

Best

ErichStyger, 2017-01-09:
No, you are absolutely correct, this is indeed a bad bug :-(
I have made a fix in https://github.com/ErichStyger/mcuoneclipse/commit/9b7eedddd8b24968128582aedc63be95b61f782c
Thanks for finding it!

closed 2017-01-09 by ErichStyger