ErichStyger/mcuoneclipse #36: Weird Problem when snapshotting components

mstroven, 2017-02-25:
Hello Erich,
I have a strange issue which arose after I made a snapshot of my projects components.  I'm using FreeRTOS 9.0.0 with Segger System Viewer and Command Line shell capabilities.  I can create the components and everything is fine.  I generate code and debug with no issues.  I can run the code and attach with system viewer and it's all good UNTIL.....

I want to get my components into version control without versioning the entire KDS3 repo.  So I have used the snapshot feature to create a repo of only the components needed by my project.  The snapshot seems to work fine, and I can check into my VCS.  However, once the project attempts to reconcile the new repository, there are a couple of problems that I can't seem to resolve.  These are with the UTIL1 component and  the CLS1 component.  Switching the FRTOS1 component back to the original repository seems to correct the issue.

Any ideas?
Best,
-Mike

ErichStyger, 2017-02-26:
Hi Mike,
I have not used the Snapshot feature, so not sure what the issue could be. What kind of problems are you seeing? If using the Snapshot feature with a local repository, have you made sure that you only using that local repository? The setting is under Windows > Preferences > Processor Expert > Repositories and Path.

ErichStyger, 2017-03-19:
Closing as not reproducible on my side (or not enough information).

closed 2017-03-19 by ErichStyger