ErichStyger/mcuoneclipse #42: PPAGE handling in FreeRTOS port for HCS12X banked memory model

grzsob, 2019-02-15:
Hi Erich,
     I tried to run your FreeRTOS port on MC9 S12XEQ512 with CodeWarrior v5.1 and PE 3.06. 
I'm using banked memory model for flash and I've encountered possible issue during context switching. In interrupt from TickCntr1_Interrupt there is call to TickCntr1_OnInterrupt() function. After this call PPAGE register value is changed to the flash page where TickCntr1_OnInterrupt() is located and then portSAVE_CONTEXT macro is called. In effect wrong PPAGE value might be saved and then restored. 
In my case issue was fixed by moving portSAVE_CONTEXT macro earlier - to ISR, before call to the TickCntr1_OnInterrupt(). 
I've checked project settings with example provided by you and they are similar (ie compiler flags that may influence this behaviour).
I'm fairly new to FreeRTOS and S12 family so it's possible that this issue is caused by something else and port unrelated. BTW thanks for all the projects and instructions you share, they all have been incredibly helpful in porting my application and using this IDE.

BR
Grzegorz Sobczyk

ErichStyger, 2019-02-15:
Hi Grzegorz,
thank you for your feedback. I have do admit that I'm not really using S12X any more, but I should still have a board somewhere. And I have used FreeRTOS with S12X only in a small project which was not banked (I believe), so it very well could be something I have missed. From your description it sounds that the Processor Expert code dealing with the tick interrupt should be different, and not sure what I have to change.
Could you provide the pice of source code you have changed so I can follow up on this? Maybe best with clear indication what you have changed where?

Thanks,
Erich

grzsob, 2019-02-18:
Hi Erich,
I attach 2 modified and 2 original files, both generated by PE. Here's what happens in my understanding:
- TickCntr1_Interrupt occurs and program goes to ISR handler, PPAGE register stays intact
- ISR handler calls TickCntr1_OnInterrupt() with CALL instruction and PPAGE register may change if there are multiple flash pages used and this function is on different one than function when interrupt happened.
- portSAVE_CONTEXT macro is called and possibly incorrect PPAGE register is pushed on stack
- at the end of function portRESTORE_CONTEXT macro is called and possibly incorrect PPAGE value is pulled from stack and copied to register which results in jumping to random place in code

Solution would be to move portSAVE_CONTEXT macro to primary ISR handler - before function call(attached files), or move everything to ISR handler (and then remove code that fixes stack with LEAS instructions).

[S12X_issue42.zip](https://github.com/ErichStyger/mcuoneclipse/files/2874503/S12X_issue42.zip)

BR
Grzegorz Sobczyk

ErichStyger, 2019-02-25:
Hi Grzegorz,
thanks for the extra details. I wanted to investigate this last weekend, but did not find the time. I hope because you have found a solution it is not that urgent?

Thanks,
Erich

grzsob, 2019-02-25:
Hi Erich,
 Of course it's not urgent, everything works fine with present
workaround. The only downside is that it has to be applied each time
after every freertos code generation (obviously). 

Best Regards 

Grzegorz Sobczyk 

W dniu 2019-02-25 07:46, Erich Styger napisał(a): 

> Hi Grzegorz,
> thanks for the extra details. I wanted to investigate this last weekend, but did not find the time. I hope because you have found a solution it is not that urgent? 
> 
> Thanks,
> Erich 

 

Links:
------
[1]
https://github.com/ErichStyger/mcuoneclipse/issues/42#issuecomment-466891146
[2]
https://github.com/

ErichStyger, 2019-02-25:
You can disable the overwrite if you disable code generation for a component, see https://mcuoneclipse.com/2012/03/23/disable-my-code-generation/

ErichStyger, 2022-01-25:
closing issue: for now disabling code generation should do it.

closed 2022-01-25 by ErichStyger