ErichStyger/mcuoneclipse #43: could you help me i meet a error when i use this script

DerZc, 2019-05-09:
java.lang.IllegalStateException: The platform metadata area could not be written: E:\MCUXpresso\workspace" -build frdmk66f_enet__enet_txrx_transfer\.metadata.  By default the platform writes its content
under the current working directory when the platform is launched.  Use the -data parameter to
specify a different content area for the platform.

i want know how can i solve it

ErichStyger, 2019-05-10:
What script are you using?
it looks you might have a disk size (no space?) or a memory (not enough RAM?) problem.
It is strange that it reports frdmk66f_enet__enet_txrx_transfer.metadata as folder.
Does it happen on another machine too? Other than that, I really don't know how to help, sorry.

ErichStyger, 2022-01-25:
closing old issue due inactivity. Reopen if this one is still valid.

closed 2022-01-25 by ErichStyger